rkoshak/sensorReporter #43: Not Really an Issue

markrad, 2017-12-07:
Hi @rkoshak,

I often see you in the OpenHab community forum and I saw mention of this project in one of your posts. First off this is a great idea. I love the idea of being able to deploy a sensor with just an ini file. 

However, it did not meet all my needs. 

First, I would really appreciate it if you would integrate the DHT22 code that you have waiting.

Second, I really needed event detection for a motion sensor. I forked your code and added that functionality (for the Pi sensor only). The other issue that I had is that my original code used an LED to indicate motion. I could have pushed the value to OpenHab and then have it return a value to turn on the LED but that really didn't meet my needs. I want the LED to tell me the local code is working. Pushing it to OpenHab has too many failure points so I wouldn't know if OpenHab was down, my network connection was bad or my code monitoring the sensor was not running.

To address this, I added an option in the sensor options to dynamically load a script that is called when the pin changes state. Here is a snippet of my ini file:
```ini
[Sensor2]
; Native Raspberry Pi Library
Class: rpiGPIOSensor.rpiGPIOSensor
Pin: 12
Connection: MQTT
Destination: sensor2/topic
PUD: DOWN
EventDetection: BOTH
StateCallback: switchLed
StateCallbackArgs: Actuator_Unoccupied,Actuator_Occupied
Poll: -1
```
Here is a sample StateCallback function (switchLed.py):
```python
occupiedPin = None
unoccupiedPin = None

def init(params):
    global occupiedPin
    global unoccupiedPin
    
    pins = params('StateCallbackArgs').split(',')
    occupiedPin = pins[0]
    unoccupiedPin = pins[1]

def stateChange(state, params, actuators):
    global occupiedPin
    global unoccupiedPin
    
    print(state)
    
    if (state == 1):
        actuators[occupiedPin].on_direct_message("ON")
        actuators[unoccupiedPin].on_direct_message("OFF")
    else:
        actuators[occupiedPin].on_direct_message("OFF")
        actuators[unoccupiedPin].on_direct_message("ON")
```
In my scenario I have a RGB LED which I turn on the red LED when no motion is detected and the blue LED when motion is detected. If this is of interest to you then I will send you a pull request. You can check out the code at https://github.com/markrad/sensorReporter.

Anyway, thanks for all your hard work. It works very well.

M.

rkoshak, 2017-12-07:
I've been really swamped lately. My plan is to work down the PR and the other issues I've saved up this weekend.

I'll happily accept the PR. My only stipulation would be that the StateCallback not be required and there is some sort of checking and logging if the switchLed.py doesn't exist. 

For the other bugs, it will be a lot easier to track if you file a separate issue and PR for each one, though I'll happily accept the lot of them in one go if necessary.

rkoshak, 2017-12-07:
I went ahead and merged the DHT22 code but am keeping the issue open until the install.sh gets updated with the required libraries.

markrad, 2017-12-07:
I might be able to split the bug fixes out. I'll give it a go. I have a pr for the DHT22 module too. I added an ini option to return the temp in either C or F.

markrad, 2017-12-12:
I thought there were two bug fixes but it was only one. Toggle in the rpiactuator code would always be true. I sent you a pull request - https://github.com/rkoshak/sensorReporter/pull/44. 

Thanks

closed 2017-12-14 by markrad