rkoshak/sensorReporter #53: Q: BT detection of paired devices?

Bibo-Joshi, 2018-03-10:
Hi there!

Currently, I'm trying to detect wether a BT adapter for my stereo is online. At first it went great using an [openHAB binding](https://github.com/sputnikdev/eclipse-smarthome-bluetooth-binding) as suggested [here](https://community.openhab.org/t/setting-up-bluetooth-binding-in-2017/20479/6). Later I noticed, that the adapter will be marked as offline, if paired with my laptop. For me, this is unwanted behavior.

While searching for solutions, I found this repository (e.g. [here](https://community.openhab.org/t/setting-up-bluetooth-binding-in-2017/20479/3)). To save some time installing, configuring etc. I wanted to ask beforhand:
**Is it possible to detect BT devices wich are paired with other devices?**

I'd be grateful for a short reply.
Kind regards,

Hinrich

rkoshak, 2018-03-12:
I do not know the answer to your question. I think that it depends on the BT device and what protocols it uses. For example, a phone which can connect to multiple BT devices at the same time will certainly still be visible. I have no idea if a device like a stereo or speaker which is designed to only be connected to one other device at a time can still be seen while connected. I suspect the answer is "no" given what you've seen with the OH BT binding.

closed 2018-03-12 by rkoshak

Bibo-Joshi, 2018-03-12:
Okay. Thanks for your kind reply anyway :)