rkoshak/sensorReporter #55: [rpiGPIOActuator.py] ON and OFF msg commands do not work

bgilmer77, 2018-03-23:
Sending any message to the topic for the actuator causes a toggle if toggle is set to true.  Setting toggle to false and then sending either NO or YES to the MQTT topic causes a message such as, '2018-03-23 15:00:50,987 ERROR - Bad input <paho.mqtt.client.MQTTMessage object at 0xb66014b0>'

Problem appears to be that the IF statements such as:
'if msg == "ON''

never match.

bgilmer77, 2018-03-23:
Additional information - the variable `msg` contains values such as the following:

`<paho.mqtt.client.MQTTMessage object at 0xb659f4b0>`

I think it is supposed to contain the actual message such to the topic, such as `ON`.

rkoshak, 2018-05-24:
This should be fixed now. Please test it too and tell me whether it works or not. Thanks.

bgilmer77, 2018-05-25:
Thanks Rich!!  I have a pretty hectic travel schedule coming up, but when I get a chance to check this out I will report back.  I really appreciate your working on this. Would have fixed it myself but did not know about the '.payload'.

rkoshak, 2018-10-03:
I'm going to close this issue. If it is still a problem please reopen.

closed 2018-10-03 by rkoshak