rkoshak/sensorReporter #63: Python Thread Error

msw1970, 2018-11-22:
Hi

I've recently updated using git pull and now I'm seeing sensorReporter stopping with the following error showing in the logs..

`● sensorReporter.service - Reports status and sensor readings over MQTT and REST
   Loaded: loaded (/etc/systemd/system/sensorReporter.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-11-22 22:41:12 GMT; 10min ago
 Main PID: 4582 (python)
   CGroup: /system.slice/sensorReporter.service
           └─4582 /usr/bin/python /opt/sensorReporter/sensorReporter.py /opt/sensorReporter/sensorReporter.ini

Nov 22 22:45:23 WGB01AP1002 python[4582]: Published message 'OFF' to 'Mark_Phone_Bluetooth'
Nov 22 22:45:23 WGB01AP1002 python[4582]: Published message 'OFF' to 'Jo_Phone_Bluetooth'
Nov 22 22:45:23 WGB01AP1002 python[4582]: Traceback (most recent call last):
Nov 22 22:45:23 WGB01AP1002 python[4582]:   File "/opt/sensorReporter/sensorReporter.py", line 207, in <module>
Nov 22 22:45:23 WGB01AP1002 python[4582]:     main()
Nov 22 22:45:23 WGB01AP1002 python[4582]:   File "/opt/sensorReporter/sensorReporter.py", line 82, in main
Nov 22 22:45:23 WGB01AP1002 python[4582]:     Thread(target=check, args=(sensors[s],)).start()
Nov 22 22:45:23 WGB01AP1002 python[4582]:   File "/usr/lib/python2.7/threading.py", line 736, in start
Nov 22 22:45:23 WGB01AP1002 python[4582]:     _start_new_thread(self.__bootstrap, ())
Nov 22 22:45:23 WGB01AP1002 python[4582]: thread.error: can't start new thread
`
I"m struggling to see why this would suddenly start happening.

Any clues?

rkoshak, 2018-11-23:
I can't imagine why this would start happening. Can you post your .ini (obfuscating anything sensitive?

I just reinstalled sensorReporter using a pull from head and I'm not seeing this error. But I don't use all of the available sensors or actuators.

msw1970, 2018-11-24:
Here's the ini file... It's pretty basic, just two bluetooth sensors, logging and REST config

`[Sensor1]
Class: bluetoothScanner.btSensor
Address: 9C:E6:5E:3D:50:9A
Destination: Mark_Phone_Bluetooth
Connection: REST
; Needs to be greater than 1 if using the RSSI mode, 25 if using the LOOKUP mode.
Poll: 2
; Uses signal strength to detect presence of devices
Mode: RSSI
; RSSI - Maximum count for near/far readings
Max: 20
; RSSI - How many near counts before declaring device present
Near: 2
; RSSI - How many far counts before declaring device absent
Far: 15

[Sensor2]
Class: bluetoothScanner.btSensor
Address: CC:25:EF:43:C8:67
Destination: Jo_Phone_Bluetooth
Connection: REST
; Needs to be greater than 1 if using the RSSI mode, 25 if using the LOOKUP mode.
Poll: 2
; Uses signal strength to detet presence of devices
Mode: RSSI
; RSSI - Maximum count for near/far readings
Max: 20
; RSSI - How many near counts before declaring device present
Near: 2
; RSSI - How many far counts before declaring device absent
Far: 15

[Logging]
File: /var/log/sensorReporter.log
MaxSize: 67108864
NumFiles: 10
Syslog: NO

[Connection1]
Class: restConn.restConnection
Name: REST
URL = http://localhost:8080/rest/items/
`

rkoshak, 2018-11-24:
Do you have anything else running that uses the BT device on this machine? 

I suspect what is happening is that the threads being created to poll for the BT devices is getting stuck  waiting for access to the device. And then the threads pile up until the max thread count is reached.

Another thing to try is increase the poll to 25 just to see if that makes a difference. 

And of course, reboot the machine to clear out any lingering locks or the like.

I don't use the BT scanner sensor any more (I find the hping3 approach to work better) and it has been a long time since I looked at this code. But I do remember I started to run out of threads when I tried to use sensorReporter BT scanning at the same time as running reelyActive. reelyActive monopolized the access to the device and the sensorReporter threads just piled up.

rkoshak, 2020-08-17:
Closing as solved by #67

closed 2020-08-17 by rkoshak