rkoshak/sensorReporter #65: Heartbeat.py does not send frequent messages

dersch81, 2019-06-02:
Hi, by using /device/getUpdate i get messages back of /status/heartbeat and /status/hearbeatstr (there is a typo) but normally i should get it every hr (3600000ms) but it does not work. I always need to ask.

rkoshak, 2019-07-11:
Can you provide more information. What communicator are you using? What is the .ini config of the heartbat plugin?

dersch81, 2019-08-04:
i'm using mosquitto 

the .ini config of the heartbeat is

`[Sensor2]
Class = heartbeat.heartbeat
Type = Heartbeat
Connection = MQTT
Poll = 3600000
; The number of milliseconds sensorReporter has been online is published to this destination
Num-Dest = klopi/heartbeat
Str-Dest = klopi/hearbeatstr
`

rkoshak, 2020-08-17:
I've just done a rewrite of sensor_reporter. I have struggled to figure out why you saw the behavior you saw. Now, more than a year later I realize the problem. The Poll is seconds, not milliseconds. 

I think I've addressed this issue with the new documentation.

dersch81, 2020-08-17:
OK :) that's funny. Thank you :D

rkoshak, 2020-08-17:
Closing as solved by #67

closed 2020-08-17 by rkoshak