rkoshak/sensorReporter #66: active_low

brononius, 2019-11-13:
Would be nice to have an option to swap the ON/OFF value coming from openHAB.
Or am I missing a setting somewhere? :$

ps love this script! Works perfectly ...

rkoshak, 2019-11-13:
I'll look into this. My original assumption was that if a transform like that is required it would have been done on OH. I've learned a lot about Python programming and MQTT since I've last updated this script and plan a significant rewrite, but have no idea when I'll get to it. I started but ended up heading down the wrong direction and need to regroup and start again.

If this were easy to implement I'd do it but it will have impacts to both the toggle and the on/off behavior as well as an additional configuration parameter which I fear would be a breaking change. When I rework it, I might either make what ON means configurable, or change to accept "HIGH" and "LOW" as the MQTT messages so you can choose on the sending side what you want the pin to do.

In the mean time, you could use a MAP transform to flip the states before going out as a message.

brononius, 2019-11-13:
Thanks! 
It's not urgent, just a bit 'weird'. ;)

Somewhere, I encounter the option 'active_low'. 
Maybe this is something you can easily add? :$

rkoshak, 2019-11-13:
active_low is a parameter for reading a pin, and it's already available for the sensor. For controlling the pin, you tell it to go high or low. I've implemented it so ON == HIGH and OFF == LOW.

brononius, 2019-11-13:
Ah, I see.
I've tried the Connection part, but couldn't get it to work. My MQTT messages were not being 'translated'.  Not sure if that's the meaning of this.
 It's my first experience with raspberry, so a lot of new things (gpio tables and so).. ;)

```
[Connection1]
Name = PinConn
Class = rpiGPIOConn.rpiGPIOConn
Actuator = Actuator1
# First value set pin off and second on. Defaults to OFF,ON
Values = AAN,UIT
```

`mosquitto_pub -h 192.168.111.45 -t relay/02/cmd -m "AAN"`

rkoshak, 2019-11-13:
sensorReporter doesn't translate anything. For the rpiGpioActuator, when it receives the message "ON" it sets the GPIO pin to HIGH (i.e. applies voltage to the pin). When it receives the message "OFF" it sets the GPIO pin to LOW (i.e. removes the voltage from the pin).

The rpiGPIOConn is not something I've written but was contributed by another developer. This was created to activate or deactivate a GPIO pin when another sensor publishes something. You cannot send a message to rpiGPIOConn directly from MQTT. 

The flow for processing an MQTT message:

`Sender -> MQTT Broker -> sensorReporter MQTT Connection -> sensorReporter Actuator`

The flow for using rpiGPIOConn is:

`sensorReporter Sensor -> rpiGPIOConn`

MQTT isn't involved.

You want to use the rpiGPIOActuator.rpiGPIOActuator (Actuator1 in default.ini) to receive the MQTT message and activate/deactivate a pin based on a message. And you will need to translate the "AAN" to "ON" and "UIT" to "OFF" in openHAB before sending the message.

rkoshak, 2020-08-17:
Note, I've rewritten sensor_reporter which hopefully makes how this stuff works more clear. 

As part of this I've removed the rpiGPIOConn in favor of a new approach. I've create a "Local" connection that can be used to link Sensors and Actuators. If you need help with this on the new version let me know.

rkoshak, 2020-08-17:
I can't say I really solved this by PR #67, but I believe the problem is overcome by events and a wholly new approach is required. So I'm closing this.

closed 2020-08-17 by rkoshak