rkoshak/sensorReporter #68: Implement Homie for MQTT

rkoshak, 2020-08-17:
Implement a Homie communicator to enable automatic discovery of sensor_reporter instances.

evotodi, 2021-02-07:
I will second this as it would allow sensor discovery in Openhab.
May I suggest https://pypi.org/project/homie-spec/ for the implementation. I have found it rather easy to use.

rkoshak, 2021-02-08:
It's still on the list of things to do but not high on the list at the moment. I just went through a rewrite of sensor_reporter and I fear it will require another rewrite to support homie because it will need to know about all the sensors and actuators up front. And I'll have to add additional information to each add-on so it can describe the type and other information needed by Homie. The current architecture doesn't support that.

Thanks for the library though, I'll definitely look into that. This will get done, just not soon.

DanielDecker, 2022-03-19:
I think the discussion about sensor / actuator discover API fits better in this topic, since it is close related to homie.
>>So sensors and actuators would get an interface to query their property e. g. datatype, measurement unit.

> Or maybe it needs to be a combination of the two where the config on the actuator gets passed through to the Homie connection and that becomes our API.

I agree with that. Some homie properties are fixed for a sensor / actuator and can be hard coded. E. g. the dht_sensor.py will have the `homie name` and `unit` in the config. While the later is necessary for all connectors.
I was thinking of a new class which holds all the possible properties that are necessary e. g. for homie. While the properties should be general enough so other systems with device discovery can make use of it.
The idea is each sensor / actuator crates a child of that class and fills in known parameters. Later a connector e. h. "homie conn" can loop thru all devices and query the properties.

rkoshak, 2022-03-28:
I don't think that YAML would work. What if I have multiple actuators on the one device? Also, a lot of that information is going to be hard coded in the actuator. For example, a GPIO actuator could only ever be a boolean and the very fact that it is an actuator means that it is settable. 

I'm thinking something like the following.

There is already a step for actuators where they `register` with the connections. This is how they tell the connection which MQTT Topic/openHAB Item to monitor for commands. We could add a similar registration to sensors (which would be a noop for MQTT, openHAB and local connections) and have the actuators and sensors pass the whole config for that actuator on registration. Then the Homie connection can publish what it needs to to the relevant topics at that time for discovery.

But not everything needs to be defined in that yaml config file. Each sensor and actuator knows a bit about itself. For example a GPIO actuator can only really be some variation of a boolean or a toggle. So we don't need to have the user tell the Homie connection that through the YAML. The sensor and actuator can add that part itself to what it passes to the connection when it registers. 

Similarly I think Homie defines what the cmd and state topics need to be so we really only need the main topic and the rest can be assumed. Even if Homie doesn't specify that, we can take a stand here and create a standard that we follow. 

So the config might be something like:

```
Actuator:
    # blah blah blah
    Connections:
        Homie:
            topic: main-light
            name: Living-room main light
            type: light
```

We know it's boolean because it's a GPIO actuator.
We know the cmd and src stuff because we follow a standard for those.
We know it's settable because it makes no sense to have an actuator that can't be commanded.

DanielDecker, 2022-04-01:
>No, I'm thinking replacing that function because the current one would be irrelevant. Right now the actuator calls register on the connection passing what we currently call the cmd_src. This tells the connection "when you receive a message on this Item/MQTT Topic, pass that message to me."

>My proposal is instead of just passing the cmd_src we pass the whole YAML config plus any hard coded defaults to the connection. Then the connection can do what ever it needs to to configure the listener (and publisher?) based on that config.

I see. Currently at #97 the register function passes the connection relevant dictionary to the connection. So with this configuration:
```yaml
#definition of logging and connectors

SensorMainLightSwitch:
   Class: gpio.rpi_gpio.RpiGpioActuator
   Level: INFO
   Pin: 8
   Connections:
      local:
         Switch:
            StateDest: eg_k_greenLED
         LongButtonPress:
            StateDest: ge_k_long_press
      mqtt:
         ShortButtonPress:
            StateDest: main_switch/short_press
            Retain: yes
      openHAB:
         ShortButtonPress:
            Item: eg_k_short_press
      Homie:
         topic: main-light
         name: Living-room main light
         type: light
```
The homie connection would receive:
```
{  'topic': 'main-light',
    'name': 'Living-room main light'
    'type': 'light'  }
```
This is not the whole actuator config as you described but I think the homie connection wouldn't gain more from receiving the whole config. Is it enough if a connection receives only its sub tree form the connection section?

>I'm cautious here because I wouldn't want to hard code too much connection specific config stuff into the device classes. Right now the configs are almost just a pass through which is pretty powerful because we can add a new connection and not have to touch any other code. It's just a different config. I'd hate to lose that by adding special knowledge to the device classes that is really only relevant to the connections.

I understand you want to keep core clear of connection related stuff. But I wounder where do you want to store the hard coded device information to pass it over to the connection on registration? That's where I first thought about a new object, which was created from a class which only purpose is to store some device related data.
One solution to keep the flexibility would be that the device adds at init() the hard coded device properties to all subsections (connection name sections) of the 'Connections' section. So the homie connection would see this:
```yaml
Homie:
        topic: main-light
        name: Living-room main light
        type: light
        cmd_src:
            topic: cmd
            name: set main light
            datatype: boolean
            settable: true
        destination:
            topic: state
            name: current light state
            datatype: boolean
```
While the user has configured only:
```yaml
Homie:
        topic: main-light
        name: Living-room main light
        type: light
```
>I'd hate to lose that by adding special knowledge to the device classes that is really only relevant to the connections.

I'm not sure what you mean with special knowledge. The device classes have to know if its e. g. 'settable', while this is only relevant for some connectors like homie.

>While this approach is loosy-goosey and I wouldn't propose such in a lot of environments, I think flexibility here is more important than "good coding practices."

I agree, flexibility is more important.

rkoshak, 2022-04-01:
> Is it enough if a connection receives only its sub tree form the connection section?

Probably, unless we want the class to add stuff to it that might be relevant like whether it's a switch and other similar stuff which might be useful to fill in the stuff that Homie needs. The GPIO and exec devices are the only ones where that stuff is a little squishy so we'd need to provide more details in the config. But for sensors like Govee, each sensor is well defined and understood. If we can we should force the user to have to configure, for example that this part represent a temperature in °C and that part is % humidity and such. 

> The device classes have to know if its e. g. 'settable', while this is only relevant for some connectors like homie.

That is a problem, I agree. We need to find the middle path between pushing too much configuration on the user in cases where the device should know what it is and making too many changes that will be Homie specific. 

But we have to get that information to Homie somehow. Maybe forcing the user to have to configure it is the most flexible way to do it. Short of that we'd have to create a whole custom set of metadata that connections like Homie (in the future others?) would have to interpret and translate. 

We have the same problem even if we have an API that the connection calls on the devices instead of having this info pushed to them too.

DanielDecker, 2022-04-04:
>Probably, unless we want the class to add stuff to it that might be relevant ...

What class do you mean?

>That is a problem, I agree. We need to find the middle path between pushing too much configuration on the user in cases where the device should know what it is and making too many changes that will be Homie specific.

I agree. We could find common device properties, that are required by e. g. homie and matter and hard code them in the devices. Something which is only homie specific would be configured manually in the config.

>Maybe forcing the user to have to configure it is the most flexible way to do it.

Sure this will be very flexible and probably this could be the first step to get homie working. But I think if homie is way more complicated to configure than MQTT than many users will ignore the homie connector.

PR #101 (2023-01-30): homie connector for sensor_reporter
This draft PR implements the homie protocol with a new connector. Moreover this fixes #100, I hope you don't mind that I put this fix in this big PR.
What does this PR change in detail:
* fixed GpioSensor Parameter `PUD` and `Btn_Pressed_State` not optional, fixes #100 
* fixed some misleading comments so we can close #95 finally (the actual problem was already fixed with #97 )
* modified MQTT connector to use it as parent for the new added homie connector, fixes #68
* added interface so sensors / actuators can provide information needed for auto discovery ( [... cut]
closed 2023-01-30 by rkoshak