rkoshak/sensorReporter #70: Request to update GPIO Readme

david-r-nielsen, 2020-10-05:
Sorry if this is the wrong format for this suggestion. I could submit the proposed changes via PR is that is preferred.

The Python file rpi_gpio.py sets the GPIO mode to "BCM". A lot of the examples that I have been following use "BOARD", which caused a great deal of confusion when I was trying to figure out why the sensorReporter was logging PIN 15 as DOWN.

Consider adding a row for Pin under the table in gpio.rpi_gpio.RpiGpioSensor of the Readme file, which explains that the GPIO mode is set to "BCM" and to use the Pinout command to find the correct pin in the configuration file.

https://github.com/rkoshak/sensorReporter/blob/7c307c9e312bbc508386cf62cff7dd3e01d5a0ac/gpio/rpi_gpio.py#L28

rkoshak, 2020-10-05:
It's not the wrong forum at all. I'll gladly accept a PR with the additions. Even better would be to update the code to let the numbering be set from the .ini file. When I first wrote this six or so years ago I just picked one. I have a vague memory that the library I was using at the time only supported BCM. That library has long since been abandonded. I've just kept that way since then.

rkoshak, 2022-02-14:
Fixed through #92

closed 2022-02-14 by rkoshak