rkoshak/sensorReporter #72: GPIO Actuator always works in "TOGGLE" Mode, steady state output is not possible

darkogorgievski, 2020-12-22:
See issue title
In the readme file the TOGGLE is described as optional setting, but without it the sensor reporter logs an error.

Additionally maybe a good idea is to change the name of ÏnitialState" to "InvertOutput", which should create pulse with duration of 0,5s opposite on the "InitialState". For Example if the initial state is ON (because as in my case the relay board inputs are inverted), one might need to create pulse "ON-OFF-ON" 

Config.ini:
```
[Logging]
 Syslog = YES
 Level = DEBUG

 [Connection1]
 Class = openhab_rest.rest_conn.OpenhabREST
 Name = openHAB
 URL = http://192.168.4.227:8080
 RefreshItem = RefreshSprinklers
 Level = DEBUG

 [Connection2]
 Class = mqtt.mqtt_conn.MqttConnection
 Name = MQTTSprinklers
 Client = rPiZero
 User = ****
 Password = ****
 Host = 192.168.4.227
 Port = 1883
 Keepalive = 10
 RootTopic = sensor_reporter
 TLS = NO
 Level = DEBUG

 [Sensor1]
 Class = heartbeat.heartbeat.Heartbeat
 Connection = MQTTSprinklers
 Poll = 60
 Num-Dest = heartbeat/num
 Str-Dest = heartbeat/str

 [Actuator0]
 Class = gpio.rpi_gpio.RpiGpioActuator
 Connection = MQTTSprinklers, openHAB
 CommandSrc  = SprinklersTerraceCmd
 InitialState = ON
 Pin = 26
 Level = DEBUG

 [Actuator1]
 Class = gpio.rpi_gpio.RpiGpioActuator
 Connection = MQTTSprinklers, openHAB
 CommandSrc = SprinklersGardenHouseCmd
 InitialState = ON
 Pin = 19
 Level = DEBUG

 [Actuator2]
 Class = gpio.rpi_gpio.RpiGpioActuator
 Connection = MQTTSprinklers, openHAB
 CommandSrc = SprinklersEntranceCmd
 InitialState = ON
 Pin = 13
 Level = DEBUG

 [Actuator3]
 Class = gpio.rpi_gpio.RpiGpioActuator
 Connection = MQTTSprinklers, openHAB
 CommandSrc = SprinklersFrontCmd
 InitialState = ON
 Pin = 6
 Level = DEBUG

[... 113 lines cut ...]
Dec 21 18:59:21 raspberrypizero DEBUG - [MqttConnection] on_subscribe: Successfully subscribed <paho.mqtt.client.Client object at 0xb64e6330>, None, 11, (0,)
Dec 21 18:59:21 raspberrypizero INFO - [sensor_reporter] Creating device for Actuator7
Dec 21 18:59:21 raspberrypizero INFO - [MqttConnection] Registering for messages on 'sensor_reporter/SprinklerPiRelay8Cmd'
Dec 21 18:59:21 raspberrypizero DEBUG - [MqttConnection] on_subscribe: Successfully subscribed <paho.mqtt.client.Client object at 0xb64e6330>, None, 12, (0,)
Dec 21 18:59:21 raspberrypizero ERROR - [sensor_reporter] Error creating device Actuator7: Traceback (most recent call last):#012  File "/usr/lib/python3.7/configparser.py", line 788, in get#012    value = d[option]#012  File "/usr/lib/python3.7/collections/__init__.py", line 914, in __getitem__#012    return self.__missing__(key)            # support subclasses that define __missing__#012  File "/usr/lib/python3.7/collections/__init__.py", line 906, in __missing__#012    raise KeyError(key)#012KeyError: 'toggle'#012#012During handling of the above exception, another exception occurred:#012#012Traceback (most recent call last):#012  File "sensor_reporter.py", line 170, in create_device#012    return device(dev_conns, params)#012  File "/opt/sensor_reporter/gpio/rpi_gpio.py", line 149, in __init__#012    self.toggle = bool(params("Toggle"))#012  File "sensor_reporter.py", line 161, in <lambda>#012    params = lambda key: config.get(section, key)#012  File "/usr/lib/python3.7/configparser.py", line 791, in get#012    raise NoOptionError(option, section)#012configparser.NoOptionError: No option 'toggle' in section: 'Actuator7'
Dec 21 18:59:21 raspberrypizero DEBUG - [sensor_reporter] 0 actuators created
Dec 21 18:59:21 raspberrypizero INFO - [sensor_reporter] Creating device for Sensor1
Dec 21 18:59:21 raspberrypizero INFO - [Heartbeat] Configuing Heartbeat: msec to heartbeat/num and str to heartbeat/str with interval 60.0
Dec 21 18:59:21 raspberrypizero DEBUG - [sensor_reporter] 1 sensors created
Dec 21 18:59:21 raspberrypizero DEBUG - [sensor_reporter] Creating polling manager
Dec 21 18:59:21 raspberrypizero DEBUG - [sensor_reporter] Created, returning polling manager
Dec 21 18:59:21 raspberrypizero INFO - [PollManager] Starting polling loop
Dec 21 18:59:21 raspberrypizero DEBUG - [MqttConnection] on_publish: Successfully published message <paho.mqtt.client.Client object at 0xb64e6330>, None, 13
Dec 21 18:59:21 raspberrypizero DEBUG - [MqttConnection] Published message 18 to sensor_reporter/heartbeat/num retain=False
Dec 21 18:59:21 raspberrypizero DEBUG - [MqttConnection] on_publish: Successfully published message <paho.mqtt.client.Client object at 0xb64e6330>, None, 14
Dec 21 18:59:21 raspberrypizero DEBUG - [MqttConnection] Published message 00:00:00 to sensor_reporter/heartbeat/str retain=False
Dec 21 19:00:21 raspberrypizero DEBUG - [MqttConnection] on_publish: Successfully published message <paho.mqtt.client.Client object at 0xb64e6330>, None, 15
Dec 21 19:00:21 raspberrypizero DEBUG - [MqttConnection] Published message 60122 to sensor_reporter/heartbeat/num retain=False
Dec 21 19:00:21 raspberrypizero DEBUG - [MqttConnection] on_publish: Successfully published message <paho.mqtt.client.Client object at 0xb64e6330>, None, 16
Dec 21 19:00:21 raspberrypizero DEBUG - [MqttConnection] Published message 00:01:00 to sensor_reporter/heartbeat/str retain=False
Dec 21 19:01:21 raspberrypizero DEBUG - [MqttConnection] on_publish: Successfully published message <paho.mqtt.client.Client object at 0xb64e6330>, None, 17
Dec 21 19:01:21 raspberrypizero DEBUG - [MqttConnection] Published message 120224 to sensor_reporter/heartbeat/num retain=False
Dec 21 19:01:21 raspberrypizero DEBUG - [MqttConnection] on_publish: Successfully published message <paho.mqtt.client.Client object at 0xb64e6330>, None, 18
Dec 21 19:01:21 raspberrypizero DEBUG - [MqttConnection] Published message 00:02:00 to sensor_reporter/heartbeat/str retain=False
```

rkoshak, 2021-06-22:
I thought I had replied to this earlier. I know I ran some tests.

First, open a separate issue for the invert stuff.

The error has nothing to do with the toggle parameter being optional. I think the problem is that warning about the channel already being in use at the top prevented the rest of the constructor from running and therefore the toggle property never got set.

There is not the "Configued RpoGpuiActuator: pin..." log line I would expect to be there. I've not been able to reproduce this error so I can't test this myself.

Another possibility is that the message came in on MQTT before the actuator was done being initialized but I don't think that's the case. I'm not seeing expected log statements that would indicate that either.

DanielDecker, 2021-09-05:
I have also the problem that the actuator always toggles using the rest API.

Digging in rpi_gpio.py I found line 149 seems to have a bug.
<code>self.toggle = bool(params("Toggle"))</code>
params() returns a string and bool() will convert any string to true, except an empty string. Following this definition of [bool()](https://www.w3schools.com/python/ref_func_bool.asp).

I guess the easiest fix would be to replace `bool(params("Toggle"))` with an compare `params("Toggle").lower() == "true"`.
@rkoshak what do you think?

rkoshak, 2021-09-05:
> I guess the easiest fix would be to replace bool(params("Toggle")) with an compare params("Toggle").lower() == "true".
@rkoshak what do you think?

I'm not sure it I misunderstood something or this changed between Python 2 and Python 3. In either case, this isn't going to be the only place where a call to `bool` with a String like that is going to pop up. It'll need to be changed everywhere.

Based on my research, using https://docs.python.org/3/distutils/apiref.html#distutils.util.strtobool would be closest to my original intent. I was mainly looking for something that works similar to Ansible where you can use just about any "truthy" string and it will work. So it should be replaced with

    bool(distutils.util.strtobool(params("Toggle").lower()))

DanielDecker, 2021-09-05:
I see!
If I understand you correct some of the code was intended to be used with python2, this would explain some other errors I had.
I'll dig for the other places of `bool()` and send a PR.

rkoshak, 2021-09-06:
It was originally written in Python 2. The last rewrite moved from supporting both Python 2 and Python 3 to only support Python 3. But clearly some stuff remains. I'm limited in how much I can test  lot of this due to lack of devices and such. I do my best but bugs will always come through.

I really appreciate the PR. Thanks!

rkoshak, 2021-09-08:
Closed via PR #84

closed 2021-09-08 by rkoshak