rkoshak/sensorReporter #75: MQTT does not reconnect if the remote broker is restarted

darkogorgievski, 2021-01-09:
I have rPi Zero on which the sensorReporter is running, and the  MQTT broker is running on 2nd rPI. If the rPi running the broker is restarted, the sensorReporter doesn't reconnect:

`pi@raspberrypizero:~ $ sudo systemctl status sensor_reporter.service
● sensor_reporter.service - Reports status and sensor readings over MQTT and openHAB REST
   Loaded: loaded (/etc/systemd/system/sensor_reporter.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-01-07 14:25:26 CET; 2 days ago
 Main PID: 26660 (python3)
    Tasks: 1 (limit: 881)
   CGroup: /system.slice/sensor_reporter.service
           └─26660 /usr/bin/python3 sensor_reporter.py sensor_reporter.ini

Jan 09 23:52:12 raspberrypizero python3[26660]: WARNING - [MqttConnection] MQTT is not currently connected! Ignoring message
Jan 09 23:52:12 raspberrypizero python3[26660]: WARNING - [MqttConnection] MQTT is not currently connected! Ignoring message`

rkoshak, 2021-06-22:
I've been unable to reproduce this. When I restart the broker the sensorReporters keep trying to reconnect until there is a success. There should be logs generated when the connection disconnected. What does it say? If you set the logging to debug for the MQTT connection you should see it trying to reconnect and when it all works see it reconnect.

DanielDecker, 2021-11-28:
I also cannot reproduce this. sensor_reporter is even reconnecting properly if two MQTT connections are configured or the ethernet cable is unplugged for several minuets.
However I had some problems after the install of Mosquitto-MQTT-Server. After the install everything seemed to work, but after the first reboot Mosquitto-Service would not start because the 'mosquitto' user didn't had write access to the configured logging file.
@darkogorgievski: did you check the MQTT-Server is running properly?

mjb00, 2021-12-03:
I just had a similar problem. My network went down and when it came back up, it would not auto reconnect to the MQTT broker.

Here is from the log. The error continued until I restarted sensor reporter. The MQTT Broker was up the whole time.

2021-12-03 16:32:40,151    DEBUG - [ MqttConnection] Published message 100 to sensor_reporter/govee/GVH5075_14F8/battery retain=False
2021-12-03 16:32:40,162    DEBUG - [ MqttConnection] on_publish: Successfully published message <paho.mqtt.client.Client object at 0xb6883b70>, None, 28483
2021-12-03 16:32:40,169    DEBUG - [ MqttConnection] Published message 72.23 to sensor_reporter/govee/GVH5075_14F8/temp_f retain=False
2021-12-03 16:32:40,180    DEBUG - [ MqttConnection] on_publish: Successfully published message <paho.mqtt.client.Client object at 0xb6883b70>, None, 28484
2021-12-03 16:32:40,185    DEBUG - [ MqttConnection] Published message -50 to sensor_reporter/govee/GVH5075_53B2/rssi retain=False
2021-12-03 16:33:31,031     INFO - [ MqttConnection] Disconnected from MQTT broker with client <paho.mqtt.client.Client object at 0xb6883b70>, userdata None, and code 7
2021-12-03 16:33:31,049    ERROR - [ MqttConnection] Error puiblishing update 22.75 to sensor_reporter/govee/GVH5075_53B2/temp_c
2021-12-03 16:33:31,072  WARNING - [ MqttConnection] MQTT is not currently connected! Ignoring message
2021-12-03 16:33:31,092  WARNING - [ MqttConnection] MQTT is not currently connected! Ignoring message
2021-12-03 16:33:31,124  WARNING - [ MqttConnection] MQTT is not currently connected! Ignoring message
...
2021-12-03 17:19:36,149  WARNING - [ MqttConnection] MQTT is not currently connected! Ignoring message
2021-12-03 17:19:36,158  WARNING - [ MqttConnection] MQTT is not currently connected! Ignoring message
2021-12-03 17:19:36,166  WARNING - [ MqttConnection] MQTT is not currently connected! Ignoring message
2021-12-03 17:19:36,174  WARNING - [ MqttConnection] MQTT is not currently connected! Ignoring message
2021-12-03 17:19:36,286     INFO - [sensor_reporter] (SIGTERM/SIGINT) terminating the process: 2 <frame object at 0xb63b7c70>
2021-12-03 17:19:36,810     INFO - [    PollManager] Waiting for all the polling threads
2021-12-03 17:19:36,827     INFO - [    PollManager] Cleaning up the sensors
2021-12-03 17:19:36,848     INFO - [    GoveeSensor] Stopping Govee observer
2021-12-03 17:19:36,866     INFO - [    PollManager] Cleaning up the actuators
2021-12-03 17:19:36,876     INFO - [    PollManager] Disconnecting from connections
2021-12-03 17:19:36,884     INFO - [ MqttConnection] Disconnecting from MQTT
2021-12-03 17:19:36,892  WARNING - [ MqttConnection] MQTT is not currently connected! Ignoring message
2021-12-03 17:19:40,734     INFO - [sensor_reporter] Setting logging level to INFO
2021-12-03 17:19:40,744     INFO - [sensor_reporter] Creating connection MQTT
2021-12-03 17:19:42,939     INFO - [ MqttConnection] Initializing MQTT Connection...
2021-12-03 17:19:42,959     INFO - [ MqttConnection] Attempting to connect to MQTT broker at X.X.X.X:1883
2021-12-03 17:19:43,047     INFO - [ MqttConnection] Connection to MQTT is successful
2021-12-03 17:19:43,055     INFO - [ MqttConnection] LWT topic is sensor_reporter/status, subscribing to refresh topic sensor_reporter/refresh
2021-12-03 17:19:43,064     INFO - [ MqttConnection] Registering for messages on 'sensor_reporter/refresh'
2021-12-03 17:19:43,097    DEBUG - [ MqttConnection] Published message ONLINE to sensor_reporter/status retain=True
2021-12-03 17:19:43,106     INFO - [ MqttConnection] Connected with client <paho.mqtt.client.Client object at 0xb68ffb30>, userdata None, flags {'session present': 0}, and result code 0. Subscribing to refresh command topic sensor_reporter/refresh
2021-12-03 17:19:43,115     INFO - [sensor_reporter] Creating device for Sensor0
2021-12-03 17:19:43,132    DEBUG - [ MqttConnection] Published message ONLINE to sensor_reporter/status retain=True

DanielDecker, 2022-02-13:
@mjb00 thanks for the log. I still can not reproduce this error, but your log hints you use a different (newer) version of paho-mqtt.
`2021-12-03 16:33:31,031 INFO - [ MqttConnection] Disconnected from MQTT broker with client <paho.mqtt.client.Client object at 0xb6883b70>, userdata None, and code 7`
I never get error **code 7**. I assume this causes the trouble, since sensor_reporter trys to decode the error code but only error 1 - 5 are defined. I'll hand in a pull request which should fix the problem using the mqtt.error_string() function.
Btw. is use paho-mqtt version 1.5.1

PR #93 (2022-02-21): Reconnect to server
+added passive reconnect feature to openHAB REST (this will renew the subscription on OH if it is broken), tested on OH2 & 3
*fixed unexpected exception after MQTT connection loss on paho-mqtt v1.6.1 due to error code 7, tested on paho-mqtt v1.5.1 & v1.6.1 fixes #75
*fixed publishing of msg on expired connection after cfg reload
*removed unused import (ssl)
*small pylint fixes

As usual I tested the changes in detail.
closed 2022-02-21 by rkoshak