rkoshak/sensorReporter #76: Update openHABRestAPI to work with OH 3's authentication

rkoshak, 2021-01-13:


DanielDecker, 2021-09-05:
For what version of OH was the REST API written?

I've set up sensorReporter with OH 3.1. Obviously I had some troubles since HTTP put events resulted in HTTP error 415 and actuators ignoring value changes. I fixed these two problems in the rest_conn.py for OH 3.1 but the changes probably break the compatibility with older openHAB versions.
I'm beginner with python and OH, so I was wondering are there different authentication modes for OH REST API?

rkoshak, 2021-09-05:
I've not touched sensorReporter since before OH 3.0 was released so it will not work with the authentication build into the access of OH 3's REST API. OH 2.x had no authentication at all and OH 1.x had no REST API at all.

The actual REST API calls are all the same, it's just the authentication that is missing for OH 3 support. I'd probably add a flag to indicate whether OH 2 or OH 3 is being used and of course BasicAuth and Token Auth would need to be supported when connecting to OH 3's REST API.

I've just not had time to deal with that. In my system I use MQTT instead of the REST API so this has not been a high priority for me. But I'm always willing to accept PRs. :-)

DanielDecker, 2021-09-12:
I see! Since I'd like to use the REST API I updated the code to work with OH3.
I just tested a small commit which enabled the authentication with the API token I've created using the OH web interface.
Currently I store the API token in the config file, next to the openHAB version. But storing secrets in a plain text file feels wrong.
What would be the best practice for securely storing the API token?

rkoshak, 2021-09-13:
Managing secrets are a whole big can of worms that is hard to get right. I'm sure there are third party libraries and such that can implement that. 

I've not looked into that before (MQTT has the same issue) because:

- sensor_reporter should be running as a limited user
- the .ini file should be owned by that limited user with only user permissions for read/write

As long as you maintain physical control of the machine and you don't expose it to the internet you are reasonably safe. Only someone who is already on your sensor_reporter machine would even know that the file exists and they'd need root to access that file in the first place. At that point, there are plenty of other ways they can access your OH instance. In short, the risk is pretty low over all and up to now has not been worth the effort to implement an alternative. 

If you want to explore and implement an alternative I'd gladly accept it. But to me it's just not a high priority.

DanielDecker, 2021-09-14:
I found the python keyring library which solves the problem for desktop computes, but is useless for headless systems, since it will request to unlock the keyring if no user is logged in.
After some recherche on the topic I also think there is no easy solution to protect the password / API token against physical access. Any solution would store the key to the encrypted password / token on the same device only making it harder to setup / manage it for a user. So I have to make sure the RPi itself is save.
Thanks for your explanation!

PR #86 (2021-10-05): OpenHAB Token-API#2
exec actor supports switch, integer items after all -> updated the readme
put every sentence in a own line in open_hab readme

includes #85 
fixes #76
closed 2021-10-05 by rkoshak