rkoshak/sensorReporter #83: Bad Command Actuator

computeq, 2021-06-22:
Actuator 0 does not work for me . If Toggle is left TRUE system toggles the actuator like expected. If i set FALSE for the toggle setting it still toggles regardless of the command sent to the topic. If i leave the Toggle setting blank the logs show that the toggle setting is set to false but i get the error Bad command ON and or Bad command OFF when trying to send either or to the topic.

[Logging]
Syslog = YES
Level = INFO

[Connection1]
Class = mqtt.mqtt_conn.MqttConnection
Name = MQTT
Client = Fishtank
User = xxxxxxx
Password = xxxxxx
Host = x.x.x.x
Port = 1883
Keepalive = 10
RootTopic = fishtank
TLS = NO
Level = DEBUG

[Actuator0]
Class = gpio.rpi_gpio.RpiGpioActuator
Connection = MQTT
CommandSrc = light/switch
Pin = 24
InitialState = OFF
Toggle =
Level = DEBUG

Some logs
root@fishtank:/opt/sensorReporter# systemctl restart sensor_reporter.service 
root@fishtank:/opt/sensorReporter# journalctl -f -u sensor_reporter.service 
-- Logs begin at Tue 2021-06-22 15:34:52 BST. --
Jun 22 17:30:22 fishtank python3[12010]: INFO - [sensor_reporter] Creating device for Actuator0
Jun 22 17:30:22 fishtank python3[12010]: DEBUG - [MqttConnection] on_subscribe: Successfully subscribed <paho.mqtt.client.Client object at 0x76588f90>, None, 1, (0,)
Jun 22 17:30:22 fishtank python3[12010]: INFO - [MqttConnection] Registering for messages on 'fishtank/light/switch'
Jun 22 17:30:22 fishtank python3[12010]: INFO - [RpiGpioActuator] Configued RpoGpuiActuator: pin 24 on destination light/switch with toggle False
Jun 22 17:30:22 fishtank python3[12010]: INFO - [PollManager] Starting polling loop
Jun 22 17:30:22 fishtank python3[12010]: DEBUG - [MqttConnection] on_subscribe: Successfully subscribed <paho.mqtt.client.Client object at 0x76588f90>, None, 4, (0,)
Jun 22 17:30:22 fishtank python3[12010]: DEBUG - [MqttConnection] on_subscribe: Successfully subscribed <paho.mqtt.client.Client object at 0x76588f90>, None, 5, (0,)
Jun 22 17:30:22 fishtank python3[12010]: DEBUG - [MqttConnection] Received message client <paho.mqtt.client.Client object at 0x76588f90> userdata None and msg <paho.mqtt.client.MQTTMessage object at 0x760926f0>
Jun 22 17:30:22 fishtank python3[12010]: INFO - [RpiGpioActuator] Received command on light/switch: OFF
                                          Toggle = False Pin = 24
Jun 22 17:30:22 fishtank python3[12010]: ERROR - [RpiGpioActuator] Bad command OFF
Jun 22 17:30:52 fishtank python3[12010]: DEBUG - [MqttConnection] Received message client <paho.mqtt.client.Client object at 0x76588f90> userdata None and msg <paho.mqtt.client.MQTTMessage object at 0x760926f0>
Jun 22 17:30:52 fishtank python3[12010]: INFO - [RpiGpioActuator] Received command on light/switch: ON
                                          Toggle = False Pin = 24
Jun 22 17:30:52 fishtank python3[12010]: ERROR - [RpiGpioActuator] Bad command ON
Jun 22 17:31:00 fishtank python3[12010]: DEBUG - [MqttConnection] Received message client <paho.mqtt.client.Client object at 0x76588f90> userdata None and msg <paho.mqtt.client.MQTTMessage object at 0x760926f0>
Jun 22 17:31:00 fishtank python3[12010]: INFO - [RpiGpioActuator] Received command on light/switch: ON
                                          Toggle = False Pin = 24
Jun 22 17:31:00 fishtank python3[12010]: ERROR - [RpiGpioActuator] Bad command ON
Jun 22 17:31:05 fishtank python3[12010]: DEBUG - [MqttConnection] Received message client <paho.mqtt.client.Client object at 0x76588f90> userdata None and msg <paho.mqtt.client.MQTTMessage object at 0x760926f0>
Jun 22 17:31:05 fishtank python3[12010]: INFO - [RpiGpioActuator] Received command on light/switch: ON
                                          Toggle = False Pin = 24
Jun 22 17:31:05 fishtank python3[12010]: ERROR - [RpiGpioActuator] Bad command ON

Any thoughts?

rkoshak, 2021-06-22:
I think I solved this but I don't use the actuator in this way in my system. Please let me know if this fixes it for you.

computeq, 2021-06-22:
We spoke years ago on the HAB fourms and ive been using an older version of this for years with great success minus a few minor bugs so i thought i would check in and see what changes you have made over the years. I like what i see ! :D Im sort of new to the whole github process when you say you fixed this do i need to re clone the entire repository to get the fix?

rkoshak, 2021-06-22:
Yes, I made the change so you'll just need to run `git pull` where you previously cloned the repo and it should show you that one file changed. I only changed one line in one file.

computeq, 2021-06-22:
Thank you I did update the file shown below. 

 # Turn ON/OFF based on the message.
        else:  
            out = None
            if msg == "ON":
                out = GPIO.HIGH
            elif msg == "OFF":
                out = GPIO.LOW

            if out == None:
                self.log.error("Bad command %s", msg)
            else:
                self.log.info("Setting pin %d to %s", self.pin,
                              "HIGH" if out == GPIO.HIGH else "LOW")
                GPIO.output(self.pin, out)

However , 
right at start up of the service and when i try and issue an ON or OFF i get the same error referenced below. 

root@fishtank:/opt/sensorReporter# systemctl restart sensor_reporter.service 
root@fishtank:/opt/sensorReporter# journalctl -f -u sensor_reporter.service
-- Logs begin at Tue 2021-06-22 18:11:19 BST. --
Jun 22 22:33:21 fishtank python3[8732]: DEBUG - [MqttConnection] on_publish: Successfully published message <paho.mqtt.client.Client object at 0x764eafb0>, None, 3
Jun 22 22:33:21 fishtank python3[8732]: DEBUG - [MqttConnection] on_subscribe: Successfully subscribed <paho.mqtt.client.Client object at 0x764eafb0>, None, 1, (0,)
Jun 22 22:33:21 fishtank python3[8732]: INFO - [MqttConnection] Registering for messages on 'fishtank/light/switch'
Jun 22 22:33:21 fishtank python3[8732]: INFO - [RpiGpioActuator] Configued RpoGpuiActuator: pin 24 on destination light/switch with toggle False
Jun 22 22:33:21 fishtank python3[8732]: INFO - [PollManager] Starting polling loop
Jun 22 22:33:21 fishtank python3[8732]: DEBUG - [MqttConnection] on_subscribe: Successfully subscribed <paho.mqtt.client.Client object at 0x764eafb0>, None, 4, (0,)
Jun 22 22:33:21 fishtank python3[8732]: DEBUG - [MqttConnection] on_subscribe: Successfully subscribed <paho.mqtt.client.Client object at 0x764eafb0>, None, 5, (0,)
Jun 22 22:33:21 fishtank python3[8732]: DEBUG - [MqttConnection] Received message client <paho.mqtt.client.Client object at 0x764eafb0> userdata None and msg <paho.mqtt.client.MQTTMessage object at 0x75ff36f0>
Jun 22 22:33:21 fishtank python3[8732]: INFO - [RpiGpioActuator] Received command on light/switch: OFF
                                         Toggle = False Pin = 24
Jun 22 22:33:21 fishtank python3[8732]: ERROR - [RpiGpioActuator] Bad command OFF

rkoshak, 2021-06-22:
Hmmmm, this makes no sense. `out` gets initialized to None. If the message is "OFF" as seems to be indicated by the logs out should get set to `GPIO.LOW` but that's not happening. Use code fences for the logs (just like in the forum with three ` on the line before and after the logs).

Maybe the message isn't "OFF" but "OFF " (space after) or "OFF
" (newline after) or something like that. The code requires it to be exactly "OFF" with no white space or other special characters.

computeq, 2021-06-22:
Im so confused .. 
I tried everything you said and nothing worked. So i decided to play with qos and changed it to 2 ran it again and now it works? ... which makes no sense at all because even when i change it back to 0 it still works even after a service restart... Also one annoying things but not the end of the world i forgot that these relays i have been buying work in reverse lol a LOW signal energizes the relay and a HIGH turns off the relay. I should be able to just swap out the values in the code block we were editing and it would fix that .

rkoshak, 2021-06-22:
The QOS shouldn't have changed anything. I can't explain why it works now, but since it seems to work I'll close the issue.

If it comes back, double check that the message is just "ON" and "OFF" with nothing else. Reopen if the problem comes back and you have verified the messages are correct.

closed 2021-06-22 by rkoshak