rkoshak/sensorReporter #87: connect mutliple sensor_reporter via mqtt

DanielDecker, 2021-10-01:
Connecting eg. two raspberry pi's via sensor_reporter and mqtt seams to be useful for low level automation. Eg. sensor1 on pi A triggers actor2 on pi B. As I understand it using openHAB to translate such simple actions is a bad idea. So I was wondering if using mqtt for such cases is a better solution?

Btw. right now sensor_reporter logs an error when mapping an sensor to an actor via mqtt (open/close doesn't match on/off obviously)

rkoshak, 2021-10-01:
I wouldn't say it's a bad idea for OH to translate them, but I never intended sensor_reporter to be strictly tied to OH. And I'm not sure I would call an automation that involves multiple machines as being low level. Typically low level would be something that takes place on the same device, such as turning on an LED when a temp sensor is above a threshold. 

I think you could handle this with the structure of sensor_reporter as is. I'd have to see what you've configured to understand the error, but one way to implement it would be to use the local communicator and the exec actuator which calls mosquitto_pub to publish the message for the other sensor_reporter.

DanielDecker, 2021-10-03:
I agree the local connection is the best choice if available. 

Meanwhile I found my error in the config, for testing propose a sensor and actor (on the same device) are using MQTT-connection:
```
[Logging]
Syslog = YES
Level = INFO

[Connection_mqtt]
Class = mqtt.mqtt_conn.MqttConnection
Name = mqtt
Client = sensor_reporter_kitchen
User = user
Password = password
Host = ip
Port = 1883
Keepalive = 10
RootTopic = sr_kitchen

[Sensor1]
Class = gpio.rpi_gpio.RpiGpioSensor
Connection = mqtt
Pin = 8
PUD = DOWN
EventDetection = BOTH
Destination = backDoor

[Actuator_greenLED]
Class = gpio.rpi_gpio.RpiGpioActuator
Connection = mqtt
CommandSrc = backDoor
Pin = 6
InitialState = ON
```
The problem was Sensor1 sends CLOSED / OPEN to the topic backDoor while Actuator_greenLED expects ON / OFF. Digging in the sensor_reporter code I found a param not mentioned in the readme: Values
So I add following to Sensor1 which solved my problem:
`Values = ON,OFF`

Thanks for pointing out the solution with the command line tool `mosquitto_pub`.

closed 2021-10-03 by DanielDecker