rkoshak/sensorReporter #88: add support for sensor pushbuttons

DanielDecker, 2021-10-01:
As I understand it the standard use case for a sensor in this project is a switch which stays on or off. In a smart home a light might be switched via a button or a rule. The button might be pressed for a short duration to toggle the light or for a long duration to activate a smart home scene. I'd like to implement optional param's to define location (to publish) and duration for for short and long press events so the destination doesn't need to handle that. @rkoshak are you interested in that feature?

What would be the best approach to make toggle for a actor using the local and mqtt connection possible?

rkoshak, 2021-10-01:
I'm not sure I understand. It sounds like you want to detect a long press or short press of a button connected to the RPi where sensor_reporter is running. So that would be a sensor, not an actuator. 

If you are talking about trying to remotely detect a long press or short press of a remote button, I don't see how that could ever really work all the well. The vagaries of timing with network traffic and such would mean you couldn't reliably tell the difference. This sort of thing is usually best implemented on the button itself. 

To implement a toggle in an actuator, the actuator needs to implement a toggle. The GPIO actuator implements such a feature via a configuration. You could implement or add to an actuator a toggle in response to a certain message instead.

DanielDecker, 2021-10-03:
To clarify I think about new features for the gpio.sensor and the gpio.actuator.
For the sensor I'd like to implement a shortButtonPress and a longButtonPress Event, which sensor_reporter will report to the configured connection. I agree trying to detect button press duration on a remote system eg. OH is not the best idea.

For the RPi.GPIO actuator I'd like to toggle the output. Unlike the already implemented param "Toggle" the output should toggle once eg. from low to high and stay there until another trigger occurs. I want this to work with the local connection, so a button can toggle the actuator. While a OH/MQTT connection connected to the same actuator can set the output to a defined value. 

For better understanding a small example:
We have a sensor and an actuator connected locally and to an OH instance. If the sensor reports a short button press it should toggle the actuator via local connection. A long button press is not handled locally but will trigger a rule on OH. The rule will do several things and finally switch the actuator off.

Do I understand you right, you suggest to make the actuator understand "toggle" as command? So it can handle the commends ON, OFF and TOGGLE.

rkoshak, 2021-10-04:
I've no problems with adding new features to either of these so long as they don't break the existing features. If they will break things then we need to be more deliberate. Perhaps a new sensor/actuator is required. 

I thought there was already support for a toggle command in the actuator though. If not that should not be hard to add.

PR #89 (2021-12-02): RPi sensor events
Added support for detecting press duration for the GPIO sensor.
Modified GPIO actuator to allow toggle commands, added status update for all configured connections to keep them up to date. Added filters to ignore own messages and ones which no effect (output status already set).
Updated the readme's.
Tested the changes in-depth for all possible connection types.
Fixes #88
closed 2021-12-02 by rkoshak