rkoshak/sensorReporter #91: SIGHUP results in GPIO error [bug]

DanielDecker, 2021-11-28:
Sending SIGHUP to sensor_reporter results for me in an delay (closing openHAB connection) and an GPIO error when creating actuator device. The log shows that disconnecting openHAB SSE takes about 8 minutes:
```
2021-11-28 12:54:43,666     INFO - [sensor_reporter] (SIGHUP) reading configuration: 1 <frame at 0x75876900, file '/opt/sensor_reporter/core/poll_mgr.py', line 71, code start>
2021-11-28 12:54:44,168     INFO - [    PollManager] Waiting for all the polling threads
2021-11-28 12:54:44,170     INFO - [    PollManager] Cleaning up the sensors
2021-11-28 12:54:44,172     INFO - [    PollManager] Cleaning up the actuators
2021-11-28 12:54:44,174     INFO - [    PollManager] Disconnecting from connections
2021-11-28 12:54:44,176     INFO - [    OpenhabREST] Disconnecting from openHAB SSE
2021-11-28 13:03:06,549     INFO - [ MqttConnection] Disconnecting from MQTT
```
if I comment everything in rest_conn.py > _get_messages() except the for loop, it still takes 8 minutes:
```
def _get_messages(self):
        """Blocks until stop is set to True. Loops through all the events on the
        SSE subscription and if it's a command to a registered Item, call's that
        Item's handler.
        """
        for event in self.client.events():
            # If we are stopping exit.
            if self.stop:
                return
```
I guess `self.client.events()` needs further analyses. Do you have any idea why it takes so long?
Btw. when no openHAB connection is configured no delay occurs and the connections reestablished right away.
In any case after the reconnect RPi.GPIO complains:
```
2021-11-28 13:03:06,855    ERROR - [sensor_reporter] Error creating device Actuator_redLED: Traceback (most recent call last):
  File "sensor_reporter.py", line 170, in create_device
    return device(dev_conns, params)
  File "/opt/sensor_reporter/gpio/rpi_gpio.py", line 204, in __init__
    GPIO.setup(self.pin, GPIO.OUT)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
```
Obviously this error occurs for every configured GPIO actuator. I guess this is because:
```
# Set to use BCM numbering.
GPIO.setmode(GPIO.BCM)
```
in rpi_gpio.py in line 29 is outside of any class. Would it do any harm to put it into the sensor and actor constructor?

rkoshak, 2021-12-02:
I don't really use the openHAB Connection in practice so I really have no idea why it might take that long. Is OH still generating events during this time? I could see it being possible that if there are not SSE events the `for event in self.client.events()` might block waiting for an event. As long as there is no event the loop won't run and the test for `self.stop` won't run to exit. Obviously at some point after 8 minutes it times out and eventually exits.

But that's just a guess. If that is what is going on, maybe there is a flag to set or a way to work around it in the sseclient library.

> in rpi_gpio.py in line 29 is outside of any class. Would it do any harm to put it into the sensor and actor constructor?

I don't think that would be a problem unless there is some sort of problem caused by running that command more than once. Give it a try and see. It should go into the __init__ probably for both the actuator and the sensor class too. Otherwise it won't get set before creating a sensor and the error will come back.

DanielDecker, 2021-12-04:
You're right `self.client.events()` blocks since OH doesn't generate events.
I just tested it. If I send a refresh request, via OH, after I triggered SIGHUP there is no delay and sensor_reporter will restart immediately.
I'll prepare a PR.

PR #92 (2022-02-14): improved restart routine
*fixed freeze on disconnect of the openHAB restAPI connection
*modified gpio.setup and cleanup so it works properly
+added GPIO parmeter for pin numbering mode and the expected input level
 for button press events
*modified GPIO log messages to include the new parameters
*moved GPIO sensor button press code to separate class to shrink RpiGpioSensor init function
+added example circuit diagram to the GPIO readme (ceated with GIMP)

I looked into several ways to fix the openHAB connection freeze on disconnect. The one I implemented is the simplest one. It works basically [... cut]
closed 2022-02-14 by rkoshak