rkoshak/sensorReporter #94: sensor actuator evaluation [enhancement]

DanielDecker, 2022-02-20:
I think there is a need for some kind of logic processing to trigger several actuators with multiple sensors.
Example: A room with 2 motion detectors and 3 lights, all lights should switch on if either motion detector triggers. Only if both motion detectors are off the lights should switch off.

Currently there is no way to trigger several actuators with on sensor or like in the example above connect many sensors with many actuators. I would implement this with a new pseudo actuator, which will receive the sensor states and trigger the configured actuators.
What do you think about the idea?

rkoshak, 2022-02-21:
Honestly I was ambivalent about adding the ability to link up sensors with actuators in the first place. I never really intended sensorReporter to become an automation platform unto itself. But it's there now so it's hard to say "no, this isn't a good idea." 

I worry a little about a slippery slope though. What if you want the lights to turn off five minutes after both motion detectors go off? Now we need to implement timers. Before too long we've all but replaced the capabilities of openHAB itself. 

I'm not against this idea and I like it better if we can centralize it into a single actuator instead of mucking around too much in the core classes. But at some point I think a line needs to be drawn. Some things will be better implemented in a microcontroller and others in a full home automation system like openHAB. If sensorReporter extends too far into either of those realms I worry it will become too complex to use effectively.

DanielDecker, 2022-02-28:
Thanks for you feedback.
I get your point.
The automation capabilities of openHAB are sufficient in most cases, except there is a connection problem or a problem with openHAB itself. With some local logic important things like light (see example in initial comment) can still work. That's where I would draw the line. So I will limit this pseudo actuator to local receivers.

PS: The setup I'm building will have one sensor_reporter per room, so all sensors and actuators within this room will be connected to one RPi.

DanielDecker, 2022-10-21:
solved with the merge of #97

closed 2022-10-21 by DanielDecker