rkoshak/sensorReporter #100: Btn_Pressed_State not optional when PUD is DOWN

rkoshak, 2022-10-25:
The advanced parameter for GPIO Sensor Btn_Pressed_State is not optional when PUD is set to DOWN.

DanielDecker, 2022-10-29:
I'd expect `Btn_Pressed_State` to be optional regardless of the configuration of `PUD` 
The default value for `PUD` is `DOWN` (line 95):
```python
pud = GPIO.PUD_UP if dev_cfg["PUD"] == "UP" else GPIO.PUD_DOWN
```
so it shouldn't make a difference if it is configured to `DOWN` or not present in the config.
Is there a runtime exception when configured as you describe?

DanielDecker, 2022-10-31:
I looked into this with my test system. I found 2 runtime exceptions:
* `AttributeError: 'RpiGpioSensor' object has no attribute 'pud'` Triggers when `Btn_Pressed_State` is not defined, just as you describe.
* Also if `PUD` is not in the config I get a `KeyError: 'PUD'`, as I recall PUD should be optional.

I'll fix both with my next PR!

PR #101 (2023-01-30): homie connector for sensor_reporter
This draft PR implements the homie protocol with a new connector. Moreover this fixes #100, I hope you don't mind that I put this fix in this big PR.
What does this PR change in detail:
* fixed GpioSensor Parameter `PUD` and `Btn_Pressed_State` not optional, fixes #100 
* fixed some misleading comments so we can close #95 finally (the actual problem was already fixed with #97 )
* modified MQTT connector to use it as parent for the new added homie connector, fixes #68
* added interface so sensors / actuators can provide information needed for auto discovery ( [... cut]
closed 2023-01-30 by rkoshak