rkoshak/sensorReporter #106: ds18x20 configuration

DanielDecker, 2023-06-13:
For future reference: 
I deliberately designed the Homie support that way, that the config for a sensor with single output doesn't need to specify the channel. So sensorReporter will assume that the given `StateDest` is reverencing the only available channel.
So, in my opinion, both configurations (with and without channel) are correct in this case.

_Originally posted by @DanielDecker in https://github.com/rkoshak/sensorReporter/pull/105#discussion_r1225862365_

florian-h05, 2023-06-14:
I would vote for option 1:

> change the ds18x20 code and documentation, so that no extra channel is needed in the configuration. This would be consistent with the other sensors.

> If the sensor implementation uses the trigger argument in the send command configure_device_channel must also specify the channel via output_name:

... and in the YAML config. The method docs for send and configure_device_channel should be updated to mention this.

DanielDecker, 2023-06-14:
OK, I'll prepare a PR to make this sensor consistent and update the docs for `send` and `configure_device_channel`

PR #108 (2023-07-07): ds18x20 consistency
This PR fixes #106

- changes ds18x20 sensor to be consistent with the other single output sensors
- renamed parameter for `_send()` and `_publish()` commands from 'trigger' to 'output_name' to make usage clearer. Now `_send()` , `_publish()` and '`configure_device_channel()` use the same parameter name for the same thing.
- extended the documentation and added examples to clarify usage for following methods: `_publish(), _send(), verify_connections_layout(), configure_device_channel()`
- fixed `verify_connections_layout()` 3rd parameter not being optional

I tested the cha [... cut]
closed 2023-07-07 by rkoshak