rkoshak/sensorReporter #113: Convert to a module

rkoshak, 2023-10-13:
Raspberry Pi OS bookwork has tightened down the installation of libraries and all but requires a virtualenv.

In order to install this as a module requires a restructuring of the folders and creation of some new files.

DanielDecker, 2023-11-06:
I understand that Bookworm doesn't allow to install global pip packages. But I don't understand why sensorReporter has to be a module. To my understanding modules are used to import them into other projects. Is this what you plan to do this with sensorReporter?

Can I help in any way?

rkoshak, 2023-11-06:
I think it would be cool to make sensorReporter installable via pip. But beyond that I haven't given it a whole lot of thought. I started going down this path when I hit the venv issue with bookworm but ended up going another route to get it working. This is mostly just here as a reminder to come back and see if it's feasible.

DanielDecker, 2023-11-07:
OK, thanks for the info.

rkoshak, 2024-06-03:
I'm going to mark this as solved through the installation scripts. If at a future date there is a compelling reason to make it into a module (e.g. we want to distribute through pypy) we can reopen this or create a new issue.

closed 2024-06-03 by rkoshak