rkoshak/sensorReporter #116: setup script

DanielDecker, 2023-11-06:
I've just set up Pi OS Bookworm and made a fresh install of sensorReporter. Meanwhile there are many steps required, e. g. before setting up the virtualenv I had to install `python3.11-venv` (Pi OS lite).

I think a setup.sh, that installs the dependencies, creates the virtualenv and python packages, would simplify the initial setup. Moreover I think the `requirements.txt` should be split up for each plug-in group and be moved to the corresponding folder e. g. one `requirements.txt` for all GPIO plug-ins.
@rkoshak: Do you agree with this?

rkoshak, 2023-11-06:
I didn't install venv separately I think. But it may depend on how you installed. I usually install `python3-full, python3-setuptools, and python3-pip` and venv comes with one of those. 

I don't have a problem with any of this. I'm a little bit wary of installing all the libraries for all the plug-ins so if there is a way to split up the requirements.txt in a way that `pip` can understand I'm all for it. Then the end users get to choose which ones they need.

DanielDecker, 2023-11-07:
Thanks for pointing to ansible, it looks very useful for managing multiple Raspberry Pi installations.

I noticed you moved the install directory for sensorRepoter from `/opt/` to `/svr/`, is this related with Bookworm?
For me the opt directory works just fine.

rkoshak, 2023-11-07:
Ansible is *very* useful. I use it to configure and maintain *all* of my machines. And thanks to Ansible I don't have to remember how to install each and every little thing, remember what I installed where, and rebuilding any machine is super easy (I had to rebuild all my RPis recently because of failed in place upgrades to bookworm and they were all done in an hour and most of that time was spent burning the images to the SD cards. 

As for /opt verses /srv, you can use any directory you want. Even `/mystuff` would work.

I used to try to follow the [FHS](https://www.pathname.com/fhs/pub/fhs-2.3.html) which says that site specific data should go under /srv and site specific software should be installed to /opt. 

But I have some cases where there is a mix of data and software and it's not clear where is the most appropriate and since most of the time I'm dealing with containerized software, /srv is the right place. 

It became annoying to have to remember which is the right place to look so I gradually ended up moving everything I install to /srv as most of the time in my systems /srv is more correct. It's only the rare cases (sensorReporter being one) where /opt would be more appropriate. But I'm the only one on these systems so what works for me is more important than following the standard.

I guess that's a long way to say `/opt` is more correct but I'm lazy and put everything `/srv` so I've only one place to look for stuff I've added to a machine.

DanielDecker, 2023-11-13:
Thanks for explaining this in detail.

> because of failed in place upgrades to bookworm

I also coudn't simply upgrade to bookworm and had to do a complete reinstall.

> I used to try to follow the [FHS](https://www.pathname.com/fhs/pub/fhs-2.3.html) which says that site specific data should go under /srv and site specific software should be installed to /opt.

In my research I came across similar information packed into a man-page. `man hier` will explain the folder hierarchy just like the link you posted. So I agree that `/svr` works also for sensorReporter too.
I'll write the setup script to will work in any chosen directory.

DanielDecker, 2024-03-02:
PR was merged, so this issue is solved!

closed 2024-03-02 by DanielDecker