rkoshak/sensorReporter #122: install script - raspi-config integration

DanielDecker, 2024-03-12:
Not a request, just a question. Could this be implemented as part of the install scripts? Obviously calling raspi-config isn't going to work but I think all that does is set a flag in a file somewhere.

_Originally posted by @rkoshak in https://github.com/rkoshak/sensorReporter/pull/121#discussion_r1519812974_

Good point, I didn't thought about that. Actually I forgot several times to enable I2C when testing the install script.
The raspi-config has actually a (non interactive mode) [https://www.raspberrypi.com/documentation/computers/configuration.html#interfacing-options-nonint] (nonint) so the i2c or 1-wire interface (Ds18x20Sensor) can be enabled or disabled with a single command.
E.g. enable i2c
`sudo raspi-config nonint do_i2c 0`

Yes it can be integrated in the install script.
Should I prepare a PR?

rkoshak, 2024-03-12:
That would be great! If you forgot I'm certain others would too. We do so much for them in the install script, it seems off to not do this if it's feasible.

DanielDecker, 2024-03-13:
> If you forgot I'm certain others would too.

So true!
I'll prepare a PR.

DanielDecker, 2024-03-21:
PR was merged, so this issue is solved!

closed 2024-03-21 by DanielDecker