rkoshak/sensorReporter #125: Kernel 6.6 breaks RpiGpioSensor EventDetection

DanielDecker, 2024-03-29:
Linux kernel 6.6 (Raspberry Pi OS - current release) breaks the RPi.GPIO event detection which is used by RpiGpioSensor.
Sources: 
* https://github.com/raspberrypi/linux/issues/6037 (using Raspberry Pi OS kernel 6.6.20-1+rpt1 on RPi4)
* https://sourceforge.net/p/raspberry-gpio-python/tickets/210/ (using CentOS7 with kernel 5.4.206-v8.1.el7 on a RPi4)

I verified this on an RPi3 running kernel 6.6.20+rpt-rpi-v7 (bookworm). Using RpiGpioSensor with EventDetection results in `RuntimeError: Failed to add edge detection`. The rest of the RPi.GPIO functionality seems to work as expected, polling a GPIO also works. With kernel 6.1 (bookworm) EventDetection works perfectly.

The RPi.GPIO project on [sourceforge](https://sourceforge.net/p/raspberry-gpio-python/code/ci/default/tree/) seems to be out of date. According to the above sources, it's still not compatible with the RPi5. Maybe we should consider a different GPIO implementation!

rkoshak, 2024-04-01:
I'm open to a different implementation. I think RPi.GPIO was originally brought in to get the edge detection. The original implementation just polled the pins every 100 msec or so IIRC. I'd hate to go back to polling but it's a lot better than not working at all.

DanielDecker, 2024-04-02:
I've looked at several GPIO libraries and decided on [lgpio](https://abyz.me.uk/lg/py_lgpio.html) [(github)](https://github.com/joan2937/lg). It's quite similar to RPi.GPIO, so it shouldn't be too hard to implement. It has all the features we use from RPi.GPIO, including edge detection. 
But it's only possible to define pins with Broadcom numbering (e.g. GPIO4), which is our default setting. So you cannot define pins with Board numbering, which is a small breaking change.

I'd start to port RpiGpioSensor and work on the other modules that relay on RPi.GPIO later on, to get RPi5 support for them too.

DanielDecker, 2024-04-14:
With PR #127 RpiGpioActuator & GpioColorLed also use the lgpio library.

So the only remaining device using the `RPi.GPIO` library is the DhtSensor. `RPi.GPIO` seams to be a sub-dependency of `adafruit-blinka` which is required by `adafruit-circuitpython-dht`. Only the latter is directly referenced by the DhtSensor.

I don't know if kernel 6.6 breaks the DhtSensor, so for now I wouldn't change the library here for now.

Since `pip install` automatically installs sub-dependencies it should be save to remove `RPi.GPIO` and `adafruit-blinka` from gpio/dependencies.txt. 
What do you think?

rkoshak, 2024-04-15:
I have to believe there are other DHT libraries out there so if adafruit is causing any problems I'm open to using something else. But since it appears not to be a problem removing RPi.GPIO and adafruit-blinka from the dependencies makes a lot of sense.

DanielDecker, 2024-04-16:
Thanks for your opinion!
In a few weeks I'll be able to test a DHT20 sensor with kernel 6.6. So you'll hear from me if there's is a problem!

DanielDecker, 2024-04-17:
This issue was solved with #127